openFyde/overlay-orangepi5-openfyde #1: In M.2 SATA writing method, stateful_partiton not found

osakanataro, 2023-01-12:
>Call sudo mount /dev/sda1 /mnt/ and echo overlays=ssd-sata | sudo tee /mnt/stateful_partition/unencrypted/Env.txt

stateful_partition not found
```
orangepi@orangepi5:~$ sudo mount /dev/sda1 /mnt
orangepi@orangepi5:~$ ls /mnt
dev_image  lost+found  unencrypted  var_overlay  vmlinuz_hd.vblock
orangepi@orangepi5:~$ ls /mnt/unencrypted/
import_extensions
orangepi@orangepi5:~$ echo overlays=ssd-sata | sudo tee /mnt/stateful_partition/unencrypted/Env.txt
tee: /mnt/stateful_partition/unencrypted/Env.txt: No such file or directory
overlays=ssd-sata
orangepi@orangepi5:~$ cat /mnt/stateful_partition/unencrypted/Env.txt
cat: /mnt/stateful_partition/unencrypted/Env.txt: No such file or directory
orangepi@orangepi5:~$
```

I tried following,  and boot successful.
```
orangepi@orangepi5:~$ echo overlays=ssd-sata | sudo tee /mnt/unencrypted/Env.txt
overlays=ssd-sata
orangepi@orangepi5:~$ cat /mnt/unencrypted/Env.txt
overlays=ssd-sata
orangepi@orangepi5:~$
```

Damenly, 2023-01-13:
Yes. Wrong instruction in release note. Thanks for pointing it out.

Damenly, 2023-01-13:
Fixed in release note.

closed 2023-01-13 by Damenly