openFyde/overlay-orangepi5-openfyde #6: Proposal for r108 script

chinen95, 2023-03-07:
Thank you for your work!
The shell used in r108 script is `#!/bin/sh`, but the code was writen for bash.
When running on Ubuntu with default settings, the following error will be reported:
`./orangepi5-openfyde-r108.run --boot sata`
`./orangepi5-openfyde-r108.run: 53: [: sata: unexpected operator`
`./orangepi5-openfyde-r108.run: 56: [: sata: unexpected operator`
`--boot requires an argument`
Because the `/bin/sh` is linked to `dash` in ubuntu by default.
Is it better to change it to `#!/bin/bash` to make it less likely that errors will occur?

Damenly, 2023-03-07:
Oh.. Indeed.  Fixed and release image was updated. It also fixes badness in MacOS because its `echo` of /bin/sh doesn't support `-n`.

chinen95, 2023-03-07:
Thank you for fixing the issue quickly!

closed 2023-03-07 by chinen95