openFyde/overlay-orangepi5-openfyde #8: r108: Couldn't complete installation.

IsQiao, 2023-03-12:
steps:
``` shell
# scp .run file to orange pi
scp orangepi5-openfyde-r108\(1\).run orangepi@192.168.1.33:~/

## uncompress the image
./orangepi5-openfyde-r108\(1\).run --boot nvme
0+1 records in
0+1 records out
4 bytes copied, 9.1256 s, 0.0 kB/s

## clear nvmen01
sudo dd bs=1M if=/dev/zero of=/dev/nvme0n1 count=2000 status=progress
sudo sync

## install 
sudo dd bs=1M if=orangepi5-openfyde-r108\(1\).img of=/dev/nvme0n1 status=progress
sudo sync
sudo poweroff

## remove tf card and reboot
echo Couldn't complete installation.
```

[image]

Damenly, 2023-03-13:
If I understand correctly, you flush the image to Nvme storage the called installation?

IsQiao, 2023-03-13:
> If I understand correctly, you flush the image to Nvme storage the called installation?

I copy the image file to nvme disk(dd command), 
then reboot, click the first button that called `install openfyde`(maybe), 
then echo the error message.

Damenly, 2023-03-13:
R108 has a small change of `how to install` from R102, the image extracted is proposed to be `dd` to TF card.
Of course, `dd` it to NVME would not influence the second button `try it` but the first button `install openfyde` won't
work because the actions which `install openfyde` do are `dd`s and file copies. Both source and target can't be same one of course.

IsQiao, 2023-03-13:
ok, i got it, i need to copy image file to TF card, reboot and install the system to nvme driver. thanks.

closed 2023-03-13 by IsQiao