openFyde/overlay-orangepi5-openfyde #10: Installing in SDCard or Pendrive

rahul-kurup, 2023-03-14:
This is not a bug per-se bug I'd like to if its possible to directly flash the OS onto a SDCard (Class A1) or a pendrive?

Damenly, 2023-03-15:
IMO current installation process doesn't cost more than directly for FreeBSD/Linux/MacOS users.
It's inconvenient for Windows users. They need to exec .run release script in orangepi5 and then copy the image to Windows back.

We'll reconsider the way of release. Thanks for your good question.

rahul-kurup, 2023-03-15:
I'm actually a Windows user. But I don't have any NVME/SSD to install this os. I have couple of SDCards and Pendrives. 
Thats why I thought to put up this question.

gustavokei, 2023-07-31:
Hi, I think the question was regarding the possibility to install the OS directly into a microSD card or pendrive.

As per [r114-r1 release](https://github.com/openFyde/overlay-orangepi5-openfyde/releases/tag/r114-r1):
> - All images generated from the .run file will support booting from a TF (microSD) card. The 'target storage' means the storage type where you want to install openFyde while running the openFyde installer.
> - You can flash the image directly to your onboard storage(SATA/NVMe/eMMC). However, you can then only choose the Try It First button at OOBE; you WILL NOT be able to perform installation under this circumstance.

From my understanding, we're supposed to install the boot img on a microSD card, and then install the system on another storage drive which doesn't include a 2nd microSD as there are only nvme, sata, emmc flags.

I think it would be a nice feature to be able to install openFyde on a second microSD or pendrive.

Damenly, 2023-08-01:
We limit installation to miscroSD cards by intention since even T0-class cards are corrupted during our IO-heavy tests. 
If you are geeks, you can dd the image generated to internal storage and then call install to microSD card : )

gustavokei, 2023-08-02:
I have managed to install openfyde into a micro sd by booting into a pendrive (img for pi5 + sata) and then installing the OS into the sd card. I guess you could do the other way around as well.

For windows users it is possible to use wsl to create your .img file.

However, wifi/bt is not working. I'm using the regular orange pi 5 with the official wifi add-on module, it seems there was [an issue in the past regarding this](https://github.com/openFyde/overlay-orangepi5-openfyde/issues/19), though I don't think it is fixed since it requires some workaround inputs from the end user and I didn't understand how to do it yet, so I'm stuck.

closed 2023-08-22 by carrie0105