openFyde/overlay-orangepi5-openfyde #13: 香橙派5 Openfyde OS 无法在NVME固态硬盘上安装

Xramyun, 2023-04-01:
[image]
在安装是无法正常安装，但可以试用
我使用的固态是梵想S500PRO
SD可以正常使用
Openfyde OS 版本为R108
[image]

[image]

[image]

Damenly, 2023-04-01:
请问你加了`—boot sata`参数了吗

Xramyun, 2023-04-01:
> 请问这个硬盘能在Orangepi5的官方系统上识别的吗？

可以

Damenly, 2023-04-01:
麻烦提供下
 按下ctrl + alt + f2，用root登录，然后输入 `lsblk` 看下输出，以及 `ls -l /mnt/stateful_partition/fyde/` `dmesg | grep -i nvme`的输出。

Xramyun, 2023-04-01:
> 麻烦提供下 按下 ctrl + alt + f2，用root登录，然后输入 看下输出，以及 的输出。`lsblk``ls -l /mnt/stateful_partition/fyde/``dmesg | grep -i nvme`

好的

Xramyun, 2023-04-02:
> 麻烦提供下 按下 ctrl + alt + f2，用root登录，然后输入 看下输出，以及 的输出。`lsblk``ls -l /mnt/stateful_partition/fyde/``dmesg | grep -i nvme`

[image]
[image]
[image]

Damenly, 2023-04-02:
你应该是直接把生成的image刷到nvme？类似https://github.com/openFyde/overlay-orangepi5-openfyde/issues/8
因为我们的预设是使用TF卡安装到NVME, 你直接在nvme试用就可以，不会影响实际体验。

Damenly, 2023-04-02:
另外这个版本开机的时候会有机率无法从NVME启动，我们会在下个版本解决该问题。

Xramyun, 2023-04-02:
> 另外这个版本开机的时候会有机率无法从NVME启动，我们会在下个版本解决该问题。

谢谢

closed 2023-04-02 by Xramyun