openFyde/overlay-orangepi5-openfyde #15: trying to upgrade to 108

333337lee, 2023-04-25:
I tried a linux device Pi4

 and the orange pi5 running openfyde 102 running Linux

@penguin    chmod +x ./orangepi5-openfyde-r108.run --boot nvme 

message unrecognized option '--boot'

will there be a 108 version we can clean install?

thank you

Damenly, 2023-04-26:
It should be:
`chmod +x ./orangepi5-openfyde-r108.run` and `./orangepi5-openfyde-r108.run --boot nvme`.
Openfyde doesn't support OTA and there will be a new release in few days please wait.

333337lee, 2023-04-26:
> It should be:
> `chmod +x ./orangepi5-openfyde-r108.run` and `./orangepi5-openfyde-r108.run --boot nvme`.
> Openfyde doesn't support OTA and there will be a new release that brings up many features and fixups in a few days please wait.

Thank you

I look forward to the next release.

just out of interest, should I have been using Openfyde 102 and running the above command in Linux?

Damenly, 2023-04-28:
> > It should be:
> > `chmod +x ./orangepi5-openfyde-r108.run` and `./orangepi5-openfyde-r108.run --boot nvme`.
> > Openfyde doesn't support OTA and there will be a new release that brings up many features and fixups in a few days please wait.
> 
> Thank you
> 
> I look forward to the next release.
> 
It's available now.

> just out of interest, should I have been using Openfyde 102 and running the above command in Linux?

If you just want the extracted .img then it's fine.

333337lee, 2023-04-28:
> It's available now.
>
>
> If you just want the extracted .img then it's fine.

thank you, the new image is also a .run file. should it be?

Damenly, 2023-04-28:
Yes, it is.

closed 2023-05-19 by Damenly