openFyde/overlay-orangepi5-openfyde #16: Transparency and white screen problem on android apps and games

zer0s1x, 2023-04-30:
After installing the aurora store, i downloaded pojavlauncher, wich is basically a minecraft java edition launcher for Android. Couldn't play because of a "transparent" glitch, where most of or the entire app would get partially or completely transparent. This happened right after i got in the game after creating a new world. Maybe it's a GPU problem... Also, in some cases, some apps would get almost completely white for some reason. Standard Notes (and pojav sometimes), wich you can download via F-Droid, is an example of this. After this, i saw leepspvideo's video about this new release, and he also got the transparency problem with the Brave browser. Here is the link for the video: https://www.youtube.com/watch?v=d6aF1HG5Hls

Here is a print of pojav's transparency problem:

[image]

Damenly, 2023-04-30:
Thanks for your report! I think it's a regression after compositing the window manager.
Our team has noticed many bugs including this one and keeps improving the Android subsystem.

zer0s1x, 2023-04-30:
Thanks! Really excited for a overall more stable version of OpenFyde for this SBC, looking forward to keeping it as my main OS!

develperbayman, 2023-07-12:
yes lots of potential also also game controllers kinda registering fyde see it but android does not xbox game controller