openFyde/overlay-orangepi5-openfyde #17: 108r can not activate gapps.

thadeustr, 2023-04-30:
108r can not activate gapps.
it says "your fydeos does not support this action..."

Damenly, 2023-05-01:
Yet another in our todo list.

Damenly, 2023-05-19:
Fixed in r108-r1 already.

dirathea, 2023-05-25:
hey @Damenly, I've tried the latest version 108r-2 and got the same issue.

[image]

[image]

Damenly, 2023-05-25:
Sorry for the wrong information. Opengapps is not supported in the version.

Damenly, 2023-07-11:
The Archero of the new version brings microG as an alternative to Gapps. You can try it.

closed 2023-07-17 by Damenly