openFyde/overlay-orangepi5-openfyde #19: Official WiFi add-on module, is it supported on OpenFyde?

vanfanel, 2023-05-09:
Hi there,

I have just booted OpenFyde on the Orange Pi 5, which has the official WiFi module add-on installed, but it's not detected on boot. Is it supported? 

Thanks!

vanfanel, 2023-05-10:
EDIT: On GNU/Linux (DietPi) it uses an overlay to work.
Adding this line to /boot/dietpiEnv.txt works:
overlays=opi5-wifi-ap6275p

I hope it helps in order to get this official add-on supported in OpenFyde.

Damenly, 2023-05-10:
There’s a mistake I made in kernel config so even you apply the dtbo it won’t work.

vanfanel, 2023-05-10:
> There’s a mistake I made in kernel config so even you apply the dtbo it won’t work.

Ah, ok, so next version will support the official WiFi add-on (ap6275p) ??

Damenly, 2023-05-10:
Yes.

closed 2023-06-08 by Damenly