openFyde/overlay-orangepi5-openfyde #20: RTL8188EUS support in OpenFyde

vanfanel, 2023-05-09:
Hi there,

I have this other WiFi USB adapter:
`
Bus 001 Device 016: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter`

which works fine on GNU/Linux, but OpenFyde doesn't seem to detect it either..
Is it supposed to work?

Thanks

Damenly, 2023-05-10:
We’ll take it into next release. Soon.

vanfanel, 2023-05-10:
@Damenly Oh thanks!

What about the official one? --> https://github.com/openFyde/overlay-orangepi5-openfyde/issues/19

I can't even get it to work on GNU/Linux.

vanfanel, 2023-05-10:
EDIT: On GNU/Linux (DietPi) it uses an overlay to work.
Adding this line to `/boot/dietpiEnv.txt` works:
`
overlays=opi5-wifi-ap6275p
`

I hope it helps in order to get this official add-on supported in OpenFyde.

vanfanel, 2023-05-10:
I see there's this already on the OpenFyde system:
`
ROOT-A/boot/rockchip/overlay/rk3588-wifi-ap6275p.dtbo`

So, the only question now is: How can I enable an overlay in OpenFyde?

Damenly, 2023-05-19:
Please try the new r108-r2.

closed 2023-06-08 by Damenly