openFyde/overlay-orangepi5-openfyde #22: MediaTek MT7610U

martinbone, 2023-05-10:
Any chance of support for the above chipset (for Archer T2U V3 wifi dongle)?

Damenly, 2023-05-10:
Does it work on armbian? And any external setup instructions?

martinbone, 2023-05-11:
> Does it work on armbian? And any external setup instructions?

Yes, it works straight away on Armbian (didn't need to install or configure anything, except the actual connection to my WiFi network). I saw another GitHub page about the chipset. Will try to find again to post here ...

martinbone, 2023-05-11:
Not sure if this is useful - but there's reference to the driver and ChromeOS:

https://github.com/sebanc/brunch/issues/513

Damenly, 2023-05-19:
Please try the new r108-r2. I don't know if it helps so expect your feedback

Damenly, 2023-05-19:
Thanks for your reply.

closed 2023-05-19 by Damenly