openFyde/overlay-orangepi5-openfyde #24: The problem seems to be with foreign language keyboard, not English keyboard.

ironiris, 2023-05-23:
Installing the Korean keyboard does not display the Korean language, only the English keyboard.
Switching between keyboard works fine.

Damenly, 2023-05-24:
Indeed a bug. We will look into this issue.

ameer4, 2023-05-27:
Same here, Arabic keyboard not functioning correctly, pressing keys only print English counterpart.

Damenly, 2023-07-10:
We have introduced new four korean input methods in https://chrome.google.com/webstore/detail/%E7%9C%9F%E6%96%87%E9%9F%B5%E8%BE%93%E5%85%A5%E6%B3%95/ppgpjbgimfloenilfemmcejiiokelkni . Please try and feedbacks are welcome.

ameer4, 2023-11-08:
> Indeed a bug. We will look into this issue.
Latest FydeOS solved the issue, can we see an update to this version?