openFyde/overlay-orangepi5-openfyde #25: Inconsistent internet access when in Linux container

martinbone, 2023-05-25:
Not sure if this is just an issue for me, and if it's a general Chromium OS issue or not ... but I find that Linux often doesn't have internet access (when accessing through terminal). Sometimes when I reboot, it does have internet access.

Damenly, 2023-05-25:
Did the host have internet access in the meanwhile? We did heavy load network tests in rk3588(s) boards without issues.

Dunno if it's a rare case or not since orange pi 5 (not  5b). My personal experience is that the connection quality of network/bluetooth via USB dongle is very bad.

martinbone, 2023-05-25:
> Did the host have internet access in the meanwhile? We did heavy load network tests in rk3588(s) boards without issues.
> 
> Dunno if it's a rare case or not since orange pi 5 (not 5b). My personal experience is that the connection quality of network/bluetooth via USB dongle is very bad.

OK. I guess it's just an issue for me (sometimes) related to my internet connection. Please feel free to close this issue. Thanks.

closed 2023-06-07 by martinbone