openFyde/overlay-orangepi5-openfyde #30: orange pi 5 sound issues

develperbayman, 2023-07-12:
sound works just not in chromium over hdmi not sure about headphone jack pluto tv android app video not working and in chromium sound not working hit or miss not sure whats going on

develperbayman, 2023-07-13:
sorry in the end i had to bail however ill be checking back in few months very interested once the some bugs have been sorted out im not on anymore back to ubuntu  ...things i noticed was sound or graphics either in android or linux and it would be hit or miss on which problem and which app ...chromium seemed to have playback issues i.e pluto tv... controller support for android apps would totally be awesome to play COD with controller... i have to also say it was the first distro that actually installed to the orange pi 5 m.2 and spi flash without issue congratulations thanks for the cool ride ill for sure be back although a little buggy id say this is in my top three opi5 distros with great potential to take the lead .....also would panfork drivers be of help in this project? i know very little of chromebook ....overall good job!

Damenly, 2023-07-17:
OK. Feel free to reopen it once you are back.

closed 2023-07-17 by Damenly