openFyde/overlay-orangepi5-openfyde #32: I still can't get foreign language keyboards to convert properly except for English.

ironiris, 2023-07-14:
Unfortunately, the Korean keyboard does not work in this version either.

No matter what keyboard I choose, it doesn't switch and I only get alphabetic input.

Damenly, 2023-07-15:
Maybe you can refer to the video? https://www.youtube.com/watch?v=jHyToDqleq8

ironiris, 2023-07-16:
> Maybe you can refer to the video? https://www.youtube.com/watch?v=jHyToDqleq8

This solved the problem, but it feels quite different from the usual Korean input method. During input, each character is spaced out, which seems to be a bug.
For the test method, if you type "dkssudgktpdy" on the Korean keyboard, it should display "안녕하세요", but it display "안 녕 하 세 요".
Of course, it's okay if the character is complete.
However, if a number is entered or the DEL key is pressed while typing Hangul, it will malfunction.
This is also a problem with the Hangul 2set input method, when I type "djqttmqslek" to create "없습니다", but the combined letters are created as "업씁니다".
And there is no such bug in the amd64 version of OpenFyde.

You can see how Korean is typed at this link.
https://ohi.pat.im/

Thank you for your kind response.