openFyde/overlay-orangepi5-openfyde #37: OPI 5 nvme SSD setup stucks on loading screen after installing to SSD from SD card.

thadeustr, 2023-07-30:
OPI 5 nvme SSD setup stucks on loading screen after installing to SSD from SD card.
I tried...
2 different sd card. A1 high speed sd cards
2 different system to compile run file to img file
Reseating nvme
Formatting nvme...
Tried with unformatted nvme
Tried with ext4 formatted nvme

Every time setup stucks on loading screen

starwolf80, 2023-08-01:
Same this is happening to me with r114-r1 on standard Orange Pi 5, using NVME boot. It's just stuck there but not frozen, since I can click around and change date/time, It just doesn't want to leave loading screen, no matter how many times I reboot or how long I wait.
Did it work on previous release?

thadeustr, 2023-08-02:
Yes. İt is not frozen. But stuck on while first name boot.
Activity led is blinking but it won't pass through loading screen.
I did install first r114 version. But now version r114 and r108 instal stucks at loading like r114-r1

starwolf80, 2023-08-02:
Not sure what's going on, I've hook it up again after few days, didn't change anything (maybe rearranged USB devices a bit) and now it boots up normally single time. Strange.

Damenly, 2023-08-03:
First boot requires loading fydeos EULA but something wrong happened on the page which blocks the page. It's fixed now.

carrie0105, 2023-08-22:
It's fixed now.

closed 2023-08-22 by carrie0105

niels1voo, 2023-09-02:
This is happening for me with 114-R1 on Orange Pi 5 plus.