openFyde/overlay-orangepi5-openfyde #40: Can't install gapps with openGapps app

mishapetrenko95, 2023-09-02:
Let.'s start small - in included fydeos store opengapps app is for x86 proccessors. 
Second, when I downloaded correct apk directly, it asked me for root permission, but included Android  subsystem wasn't rooted. 
Last, because of system wasn't rooted, I wasn't able to  reflash microG even when i downloaded different versions of it throug that app.

It's clear that there is some kind of a glitch that you can sort out. Anyway, thank you for great OS for my Orange Pi 5!!!

InfernalWolf666, 2023-10-02:
i too

xbmcnut, 2023-12-09:
> Let.'s start small - in included fydeos store opengapps app is for x86 proccessors. Second, when I downloaded correct apk directly, it asked me for root permission, but included Android subsystem [...]

Any chance you can link to the correct version so I can try too? I get this error.
[image]

NiffirgkcaJ, 2024-05-10:
I hope this issue gets remedied soon.