openFyde/overlay-orangepi5-openfyde #47: Open Fyde r120 does not boot on Orange Pi 5b emmc

Poincare1, 2024-06-09:
I follow the instructions for using the .run file and select the Pi 5b.  After some minutes, an .img file is created.  I flash this to my emmc. There are no issues or problems reported.  Upon rebooting the Orange Pi 5b, the red power led and green activity led are visible.  There is, however, no video output or anything that comes across the HDMI.  
I have tried several times, and each time with r120 the image written to the emmc does not appear to boot.

zba, 2024-09-03:
by instruction you shall flash it to sdcard  boot sd and install from there to emmc, but it not boots from sd on opi 5 plus for me too.

mhornbacher, 2024-09-26:
I don't get the green activity light. Just the red power light

justliang, 2025-01-10:
same issue as you do, I can see green led flashing, but the monitor is blank all the time. and I'm sure the bootloader is working because I can flash ubuntu and boot using the same bootloader.

> I follow the instructions for using the .run file and select the Pi 5b. After some minutes, an .img file is created. I flash this to my emmc. There are no issues or problems reported. Upon rebooting [...]

mhornbacher, 2025-01-27:
@justliang I got it working and it has been stable for months. It was a long debugging session on discord. I'll write it up when I get home and remember. Sadly my laptop was stolen so I have a lot to catch up on